simon-jouet/Marlin #1: Temperature not accurate

luc-github, 2017-12-27:
ESP32/RAMPS read temperature = -15 (not correct)
but same sensor read  24 (correct) on RE-ARM/RAMPS

simon-jouet, 2017-12-30:
So the issue is at least two fold with this.

In the board the reference voltage in the voltage divider for the temperature is the 3.3V to the microcontroller, however, the ESP32 reference voltage of the ADC is ~3.1V (varying between modules). So at low temperature the reading is capped and the temperature is wrong, but when the temperature reaches a specific temperature ~35 IIRC then it's fine because it's below the output voltage of the voltage divider is below the 3.1V reference.

The second consideration is that the ESP32 ADC is 12bits while the tables in Marlin are designed for 10bits. So either the accuracy of the ADC readings need to be lowered or the tables recomputed for 12bits. But recomputing the tables for 12bits is a bit more complicated because with the current OVERSAMPLENR the values can overflow a uint16 so the OVERSAMPLENR must be lowered

luc-github, 2018-01-13:
Ok just to confirm I have connected ESP32 sensor to existing extruder using marlin using same sensor 100K to keep same behavior   

here the results:

 Marlin temp | ESP32 temp 
-------------- | ------------ 
180 C | 125 C 
190 C | 140 C 
200 C | 150 C
210 C | 172 C
220 C | 195 C
230 C | 224 C
240 C | 260 C

so from usage point of view selecting 200 to 240 degres from ESP32 will actually heat almost same in reality => ~ 230 C

luc-github, 2018-01-13:
More measures with ADC : 

The 12 bits limits 4095 is reached under 20 degres C

ADC | ESP temp | Temp
-- | -- | --
4095 | -15 | 10
4095 | -15 | 18
3669,57 | -15 | 40
3438,71 | -15 | 50
3210,71 | -15 | 60
2902 | -15 | 70
2634 | -15 | 80
2340 | -15 | 90
2056 | -15 | 100
1774 | -15 | 110
1489 | -15 | 120
1233 | -15 | 130
1008 | 0 | 140
818 | 67,07 | 150
655,71 | 91,25 | 160
525,29 | 108,75 | 170
405,71 | 125,83 | 180
320 | 140 | 190
240 | 156,19 | 200
178,14 | 172,77 | 210
129,14 | 190,84 | 220
84,57 | 214,64 | 230
48 | 250 | 240
22,71 | 300,71 | 250
1,71 | 353,21 | 260

simon-jouet, 2018-01-13:
Hey @luc-github 

Thanks for that, if it helps the changes that I did in this branch for the temperature to ~work are:

in thermistors.h I changed `#define OVERSAMPLENR 16` (to 7) in order to use 12bit accuracy on the ADC and still have a value that fits within a short for the temperature table. Without that the 16 readings will sum up to more than a short and overflow.

And then I modified `temptable_999` to match the ADC value. This was quite a painful process of connecting an arduino with accurate temperature readings, adding a gcode to print on the console the adc value and iterating over all the temps. I don't think the table I got is very accurate and considering the effort it's probably not very practical.

I was considering writing a bit of code to convert the value into an Arduino like value (10 bits with 5V ref voltage) so each reading might be slightly more complicated but at least the current temptables should be compatible. (we will still have the issue of the low temp readings not working)

luc-github, 2018-01-14:
Hi Simon,
Convert 12 bits to 10bits is not the issue I think, there are several ways to do this easily, Due, [LPC1768](https://github.com/simon-jouet/Marlin/blob/esp32/Marlin/src/HAL/HAL_LPC1768/HAL.cpp#L159-L171) and [STM32](https://github.com/simon-jouet/Marlin/blob/esp32/Marlin/src/HAL/HAL_STM32F1/HAL_Stm32f1.cpp#L130-L132) are 12bits ADC and have working solution without modifying the sensor table,  Arduino API has also a function for this but I am not sure it is in ESP32 core:  [AnalogReadResolution](https://www.arduino.cc/en/Reference.AnalogReadResolution)
I will revert the `#define OVERSAMPLENR 7` and apply same change as Stm32 for resolution

Actually my concern is more for room temperature:  if ADCoutput is identical from 10 degres and lower to 20 degres : reference is confusing.

I did not digged in code so much yet - but will feedback soon

luc-github, 2018-01-14:
@simon-jouet in addition to discussion check the issue tracker : https://github.com/espressif/esp-idf/issues/164 which confirms there is definitly several issues with ADC and vref, unfortunatly I see no real time frame and issue is open for a while.

Reading at last comment suggesting using external ADC and considering ADC is critical in such project - what do thing about this ? based on your comment that you want to do an all in one board to optimize pins according our discussion instead of ramps -  Using external would save a lot of energy trying to solve a problem from ESP32 core/design that may not give good results

What do you think ?

simon-jouet, 2018-01-18:
Hi @luc-github 

I think you might be right about using an external ADC I thought about it too quite a bit but i'm a bit conflicted. On one hand the ESP32 contains enough for printing on the other hand like you say we could potentially save quite a bit of time.

I spent a bit of time yesterday and today giving it another short. You can have a look at the ADC branch, it doesn't really solve the room temperatures issue but the new way should be more accurate as it uses the esp-idf calibration curve and convert the value in a reading that can be used with the existing tables. I gave it a very quick shot and it looked to be quite reasonable.

There is still the V_REF that can be calibrated to give more accurate results (http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/adc.html#example-of-reading-calibrated-values) in the test i've done I left it to 1100 which is ballpark I might try to change it to the accurate value but I need a proper way to test that out. (I tried putting two thermistors on the same heater block but a small difference in position can change quite a bit the temperature)

luc-github, 2018-01-19:
I will test ADC branch and report this week end.
My suggestion of external ADC was based on my personnal experience having a reliable prototype then improve it using the solid basis. But I am fine going deep in ADC first - in both case it is fun and can learn things ^_^

Exprimental temperature measurement outside a controled lab is always suject to variation, including the hardware itself,  and if delta/tolerance is 1 or 2 or even 5 degres it is not a big issue IMHO.

luc-github, 2018-01-21:
Hi Simon - just to clarify your test using new ADC code was using the `#define OVERSAMPLENR 16` or `7` ?
we should stick to `16` right ?

simon-jouet, 2018-01-21:
I tried with the old value of 7, but it shouldn't matter because the table I was using (13) is a multiplier based on the OVERSAMPLENR. But yeah we should go back to 16, I just pushed a merge with latest Marlin 2.0 and reverted it to 16

simon-jouet, 2018-01-21:
Just FYI, I haven't pushed the ADC changes to master yet so if it's better for you the new code I can merge that in :) (or you can I think you have push access)

luc-github, 2018-01-21:
Yes saw it - no worry - I test ADC branch with OVERSAMPLENR 16 first - to have reference then will test new esp32 branch + adc change
Ok I will push commit it is ok

luc-github, 2018-01-21:
ADC branch using OVERSAMPLENR 16 and latest merge from Marlin give me same data

Temp ESP | ADC | Temp
-- | -- | --
31,67 | 962 | 19
36,33 | 950 | 30
44,39 | 924,06 | 40
54,89 | 881,5 | 50
64,7 | 831,6 | 60
75,13 | 769,11 | 70
85,15 | 700,87 | 80
95,07 | 627,5 | 90
104,42 | 557,44 | 100
114,81 | 481,37 | 110
125 | 441 | 120
135,54 | 345 | 130
145,76 | 289,19 | 140
155,7 | 242,06 | 150
165,85 | 201,87 | 160
175,67 | 169 | 170
186,8 | 138,69 | 180
196,16 | 117,44 | 190
206,91 | 96,94 | 200
217,29 | 81,25 | 210
226,81 | 69,19 | 220
238,75 | 57,25 | 230
250,08 | 47,94 | 240
260,42 | 40,75 | 250
272,5 | 34 | 260

simon-jouet, 2018-01-21:
Great so probably by tuning the VREF most of the error will be gone (or at least we will be close enough)

simon-jouet, 2018-01-21:
With the MR in we should clean this up a bit to not have the pin numbers/channels hardcoded

`get_channel(int pin)` only contains the 2 pins currently used (I didn't find an equivalent function in esp-idf). And `HAL_adc_init()` only init the ADC on hardcoded pins

luc-github, 2018-01-21:
instead of `36` use `A0`, instead of 39 use `A3`
so looking at https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/adc_channel.h

a macro should do the trick 
A0 -> ADC1_CHANNEL_0
A3 -> ADC1_CHANNEL_3

so in pins definition use chanel instead of real pin 0 and 3 and the pins A0 and A3 can be created using macro ans same for chanel

No ?

luc-github, 2019-04-03:
closed as now merged to Marlin

closed 2019-04-03 by luc-github