simon-jouet/Marlin #2: Print over serial crash

luc-github, 2017-12-28:
Simon you raised that print over serial crashed

I did not built a full system yet just your board and ramps connected by USB to my computer and using repetier host

I have remove temperature GCODE as I do not have any heater and launch a cube 20mmX20mmX20mm
It sent commands during more than 16 min for a total of 4450 commands and still responding 

09:31:30.679 : Printjob finished at 28/12/2017 09:31
09:31:30.679 : Printing Time: 16m:32s
09:31:30.679 : Lines Send: 4260
09:31:30.690 : N4450 M84*42

So like this I cannot reproduce crash 

Per my experience ESP is sensitive to power supply could it be the root cause

Do you board is able same as me at least ? not powered using external  external power supply and not using motors / heaters

I include my test file

[cube.zip](https://github.com/simon-jouet/Marlin/files/1590825/cube.zip)

simon-jouet, 2018-01-13:
I just received a new set of NEMA's so I will setup a small environment to test this out with the motors and the thermistors. I'm wondering if the issue is not to do with the temperature timer (I remember hacking quite a bit to get this working) anyway I will try with and without heaters and see if that changes anything

luc-github, 2018-01-14:
I am building a printer with swichable casing to test with several boards - so I did not tested yet with steppers in my side will feedback when done

simon-jouet, 2018-01-21:
Did some changes locally to shadow the registers for I2C following the error in #3 and it seems like it's working. (the shadowing code probably broke the endstops, I will have to try that)

I've got an extruder + 5 motors with no part cooling fan or BED and managed to complete a ~30 minutes print.

[image]

luc-github, 2018-01-21:
Nice

Z1 +Z2 did you think it is relevant ? Did you tested E1 with E0 ?
I did not tested with steeper yet

simon-jouet, 2018-01-21:
No you're right if Z1 works then Z2 will work just that I had 5 motors and my current frame is a prusa i3 clone requiring two Z steppers. I haven't tried E1 and I don't think I will because based on what we said previously probably not worth having it and freeing up those pins for something else.

simon-jouet, 2018-01-21:
`Print ended at: 19:13:31 and took 0:40:29` :)

Okay next step will be to check how much the new code broke the endstops ...

simon-jouet, 2018-06-03:
Closing this for now, the serial terminal looks to be working fine and the current code doesn't use the GPIO expander

closed 2018-06-03 by simon-jouet