simon-jouet/Marlin #5: WDT Reset

luc-github, 2018-01-21:
Just reading temperature using M105 and I get 
```
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
start
echo: External Reset
 Brown out Reset
 Watchdog Reset
```

I am not sure what trigger that as previous test on USB this did not happen - but I was was not reading temperature neither

simon-jouet, 2018-01-21:
Did you get the latest esp-idf?  I had to disable the task WDT in my config

```
$ cat sdkconfig | grep WDT
CONFIG_INT_WDT=y
CONFIG_INT_WDT_TIMEOUT_MS=300
CONFIG_INT_WDT_CHECK_CPU1=y
CONFIG_TASK_WDT=
```

luc-github, 2018-01-21:
No actually I use arduino-esp32 but disabing WDT is not safe for 3D printing - I will see where the 
 `void esp_task_wdt_feed();` should be added to avoid this

simon-jouet, 2018-01-21:
So I would *guess* the source of the problem is `Temperature::isr()` that takes too long resulting in a crash. I will try the M105 once the "printing" is done and I can check if it crashes for me or not

simon-jouet, 2018-01-21:
Okay so

```
>>>M105
SENDING:M105
ok T:130.28 /0.00 B:31.30 /0.00 @:0 B@:0
```
Enabling the task WDT

```
$ cat sdkconfig | grep WDT
CONFIG_INT_WDT=y
CONFIG_INT_WDT_TIMEOUT_MS=300
CONFIG_INT_WDT_CHECK_CPU1=y
CONFIG_TASK_WDT=y
CONFIG_TASK_WDT_PANIC=
CONFIG_TASK_WDT_TIMEOUT_S=5
CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0=y
CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=y
```

Then I get the issue even without M105 if I wait a bit I get

```
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: loopTask
```

luc-github, 2018-01-21:
yes when I add some   esp_task_wdt_feed(); I got this message instead of board reset
I remember got discussion with ESP32 developer about SPIFFS format which generated same message - but in my ESP3D I do not have such issue with WDT enable so should have a way to avoid it

luc-github, 2018-01-21:
found the root cause 
// Reset watchdog.
inline void watchdog_reset() {};
it is not defined for ESP32

luc-github, 2018-01-21:
well adding do not hurt but do not solve all reset 😢

luc-github, 2018-01-21:
Seems I did a push instead of PR sorry

luc-github, 2018-01-22:
Some progress:
Issue is in `idle();` function - will continue to dig in

luc-github, 2018-01-23:
Seems my previous post was not posted ...
WDT reset seems solved by feeding watchdog but now there is an error:
```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x40085255  PS      : 0x00060034  A0      : 0x80085843  A1      : 0x3ffc0590  
A2      : 0x3ffc112c  A3      : 0x00000000  A4      : 0x00000005  A5      : 0x00000020  
A6      : 0x00000020  A7      : 0x00060023  A8      : 0x0000cdcd  A9      : 0x0000cdcd  
A10     : 0xb33fffff  A11     : 0x0000abab  A12     : 0x00060021  A13     : 0x00000001  
A14     : 0x00060021  A15     : 0x00060023  SAR     : 0x00000017  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40085255:0x3ffc0590 0x40085840:0x3ffc05b0 0x400866c7:0x3ffc05d0 0x40086419:0x3ffc05f0 0x40081602:0x3ffc0600 0x400decdf:0x00000000
```
the stack decoding give me : 
Decoding 7 results
0x40085255: uxPortCompareSet at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4571
:  (inlined by) vPortCPUAcquireMutexIntsDisabled at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portmux_impl.h line 86
:  (inlined by) vTaskEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4216
0x40085255: uxPortCompareSet at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4571
:  (inlined by) vPortCPUAcquireMutexIntsDisabled at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portmux_impl.h line 86
:  (inlined by) vTaskEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4216
0x40085840: xTaskIncrementTick at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4571
0x400866c7: xPortSysTickHandler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./port.c line 346
0x40086419: _frxt_timer_int at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portasm.S line 303
0x40081602: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./xtensa_vectors.S line 1105
0x400decdf: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 85
```
and commenting : `void manage_inactivity(bool ignore_stepper_queue/*=false*/)` in Marlin.h make issue disapear

luc-github, 2018-01-25:
To help to narrow down issue I have commented the `HAL_STEP_TIMER_ISR` and `HAL_TEMP_TIMER_ISR` ( the one which have `Temperature::isr()` and uncomment `void manage_inactivity(bool ignore_stepper_queue/*=false*/)` and still got error but only once and little bit different :
```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU1)
Register dump:
PC      : 0x40085c0c  PS      : 0x00060034  A0      : 0x800865e1  A1      : 0x3ffc0510  
A2      : 0x3ffc1138  A3      : 0x3ffc0588  A4      : 0x00000008  A5      : 0x3ffd5158  
A6      : 0x00000017  A7      : 0x00000000  A8      : 0x0000abab  A9      : 0x0000cdcd  
A10     : 0xb33fffff  A11     : 0x0000abab  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060023  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40085c0c:0x3ffc0510 0x400865de:0x3ffc0530 0x4008515d:0x3ffc0550 0x4008450e:0x3ffc0570 0x40084459:0x3ffc05c0 0x40084a3a:0x3ffc05e0 0x400816b1:0x3ffc0610 0x40101197:0x00000000
```

```

Decoding 11 results
0x40085c0c: uxPortCompareSet at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4571
:  (inlined by) vPortCPUAcquireMutexIntsDisabled at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portmux_impl.h line 86
:  (inlined by) vTaskEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4216
0x40085c0c: uxPortCompareSet at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4571
:  (inlined by) vPortCPUAcquireMutexIntsDisabled at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portmux_impl.h line 86
:  (inlined by) vTaskEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4216
0x400865de: xTaskRemoveFromEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4571
0x4008515d: xQueueGenericSendFromISR at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1995
0x4008450e: pp_post at ?? line ?
0x40084459: lmacPostTxComplete at ?? line ?
0x40084a3a: wDev_ProcessFiq at ?? line ?
0x400816b1: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./xtensa_vectors.S line 1105
0x40101197: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 85

```
The other difference between previous test is : before I was using `WiFi.mode (WIFI_OFF);` so no wifi actifity and now I have WiFi working connected in Station mode to an Access Point
I will continue test using  `WiFi.mode (WIFI_OFF);`  because error occurrence in higher

luc-github, 2018-01-25:
Additionaly It make sense that when doing printing test there is no issue as it happen when printer is inactive `void manage_inactivity(bool ignore_stepper_queue/*=false*/)`

luc-github, 2018-07-06:
Close as isr code changed

closed 2018-07-06 by luc-github