simon-jouet/Marlin #6: DISABLED Macro conflict

luc-github, 2018-01-22:
when macro are redefined but having same meaning it can be handled but here the macro name is identical but not same meaning:
```
In file included from C:\Users\luc\AppData\Local\Temp\arduino_build_569060\sketch\src/inc/MarlinConfigPre.h:27:0,

                 from C:\Users\luc\AppData\Local\Temp\arduino_build_569060\sketch\src/inc/MarlinConfig.h:26,

                 from C:\Users\luc\Desktop\Marlin\Marlin\Marlin.ino:23:

c:\users\luc\appdata\local\temp\arduino_build_569060\sketch\src\core\macros.h:133:0: warning: "DISABLED" redefined

 #define DISABLED(b) (!_CAT(SWITCH_ENABLED_, b))

 ^

In file included from C:\Users\luc\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal.h:52:0,

                 from C:\Users\luc\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:35,

                 from C:\Users\luc\AppData\Local\Temp\arduino_build_569060\sketch\Marlin.ino.cpp:1:

C:\Users\luc\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-gpio.h:51:0: note: this is the location of the previous definition

 #define DISABLED  0x00
```

luc-github, 2018-01-23:
@simon-jouet I have just noticed : 
https://github.com/simon-jouet/Marlin/blob/esp32/Marlin/src/HAL/HAL_ESP32/HAL_ESP32.h#L10-L13
```
#undef DISABLED
#include <Arduino.h>
#undef DISABLED
#define DISABLED(b) (!_CAT(SWITCH_ENABLED_, b))
```

Which bring more confusion

I think `DISABLED` should be a value  and because `DISABLED(b)` is a test macro, it should be renamed as `IS_DISABLED(b)` which make more sense, and this  should apply to `ENABLED` offcourse for consistency

What you think ?

simon-jouet, 2018-01-23:
Hi @luc-github 

So in the original version of the HAL (a while ago) I did it that way by renaming ENABLED and DISABLED the problem is that it's used so much across the codebase that any change of this magnitude will probably result in the HAL to never been merges in the mainstream Marlin.

The issue which this bit of code is trying to address is that DISABLED is defined in the esp-idf (https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-gpio.h#L51) but it's also defined by Marlin as a test mecro to check if a pin is enabled or not. We don't use the esp-idf DISABLED so what this code does is undef the esp-idf one and redefine the Marlin one.

When I compile using arduino-esp32 as a component of the esp-idf I don't get this error (though I get a very similar one since the last update with _BV())

luc-github, 2018-01-23:
I just checked and DISABLED is not used in ESP core currently just defined so it should not bring issue now but this is potentialy source of error and hard to find  later IMHO
Did you asked feedback/advice on Marlin dev ? The Marlin 2.x is still in early stage such change is still possible in theory - I am no familiar with Marlin mindset : I know in repetier and smoothieware what kind of proposal is allowed and not but not in Marlin

simon-jouet, 2018-01-23:
I asked for that some time ago, and someone replied that it was a perfect candidate for a #undef, I also think this should be renamed as DISABLED/ENABLED are far too generic, might be worth pushing a bit more

luc-github, 2018-01-25:
Ok I will ask question again on Marlin 2.X thread - thanks ^_^

luc-github, 2019-04-03:
closed as now merged to Marlin

closed 2019-04-03 by luc-github