simon-jouet/Marlin #9: Interrupt Functions miss IRAM_ATTR attribut

luc-github, 2018-01-23:
https://github.com/espressif/esp-idf/blob/master/components/spi_flash/README.rst#iram-safe-interrupt-handlers

as I got `Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)` when I enable web server on board

I guess this must be done in HAL_timer_ESP32

luc-github, 2018-01-23:
Error reported : 
```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x400d5fa4  PS      : 0x00060034  A0      : 0x400816b0  A1      : 0x3ffc0bf0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x000000ff  A5      : 0x40086d38  
A6      : 0x00000003  A7      : 0x00060723  A8      : 0x80081238  A9      : 0x3ff5f000  
A10     : 0x3ffc109c  A11     : 0x20000000  A12     : 0x00000400  A13     : 0x3ffdf6b0  
A14     : 0x00000003  A15     : 0x00060b23  SAR     : 0x00000015  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d5fa4:0x3ffc0bf0 0x400816ad:0x3ffc0c10 0x4000bfed:0x00000000
```

```
Decoding 8 results
0x400d9d20: tempTC_Handler at C:\Users\user\AppData\Local\Temp\arduino_build_556193\sketch\src\module/temperature.cpp line 1587
0x400816b4: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./xtensa_vectors.S line 1105
0x40086d3c: _frxt_int_enter at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portasm.S line 119
0x400d9d20: tempTC_Handler at C:\Users\user\AppData\Local\Temp\arduino_build_556193\sketch\src\module/temperature.cpp line 1587
0x400816b1: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./xtensa_vectors.S line 1105
0x40081ba5: spi_flash_read at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./flash_ops.c line 508 (discriminator 1)
```

where error come from temperature.cpp
```
HAL_TEMP_TIMER_ISR {
  HAL_timer_isr_prologue(TEMP_TIMER_NUM);
  Temperature::isr();
}
```

simon-jouet, 2018-01-23:
IIRC this is dealt by the arduino-esp32 wrapper already. I'm a bit surprised this is crashing for you, it's been working fine for me (the print job I did two days ago was with the temperature of the extruder)

I had a similar kind of issue at the beginning because Marlin was using floats in the ISR, which meant that esp32 gcc was optimizing them to use the FPU, however the current toolchain doesn't deal with that properly resulting in crashes. (That's why in the adc branch I used doubles which is done in software rather than a float)

luc-github, 2018-01-23:
for what I see here : https://github.com/espressif/arduino-esp32/issues/855 this attribut must be added by  programmer what wrapper do you refer to ?

luc-github, 2018-01-24:
and if I comment the content of 
```
HAL_STEP_TIMER_ISR {/*
  HAL_timer_isr_prologue(STEP_TIMER_NUM);
  #if ENABLED(LIN_ADVANCE)
    Stepper::advance_isr_scheduler();
  #else
    Stepper::isr();
  #endif*/
}
```
and 
```
HAL_TEMP_TIMER_ISR {
 /* HAL_timer_isr_prologue(TEMP_TIMER_NUM);
  Temperature::isr();*/
}
```
the board is no more crashing when accessing web server
[image]

luc-github, 2018-03-01:
Ok I did a lot of code read and here my finding : 
1 - cli / sei do nothing they are not enabled in ESP32
2 -  CRITICAL SECTION is not enabled neither 
```
#define CRITICAL_SECTION_START cli()
#define CRITICAL_SECTION_END sei()
```` 
So I use 
```
#define CRITICAL_SECTION_START portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED; portENTER_CRITICAL(&timerMux);
#define CRITICAL_SECTION_END portEXIT_CRITICAL(&timerMux);
```

3 - In temperature.cpp : void Temperature::isr()
volatile variable `in_temp_isr` is not in CRITICAL SECTION which make tested value incorrect

4 - disable timer in isr is useless because the autoreload flag is set to false, so when arriving to isr, timer is automaticaly disabled even counter is reset by prolog function

5 - ISR function miss IRAM_ATTR which may seems bringing no issue but it is incorrect, there is an extra extern useless, and not sure why function is set as extern "C" in current case, at least no need in define 
so  
```
#define HAL_TEMP_TIMER_ISR extern "C" void tempTC_Handler(void)
#define HAL_STEP_TIMER_ISR extern "C" void stepTC_Handler(void)
extern "C" void tempTC_Handler(void);
extern "C" void stepTC_Handler(void);
````
should be  IMHO
```
#define HAL_TEMP_TIMER_ISR  void IRAM_ATTR tempTC_Handler(void)
#define HAL_STEP_TIMER_ISR  void IRAM_ATTR stepTC_Handler(void)
extern void  tempTC_Handler(void);
extern void  stepTC_Handler(void);
```
I do not have any complain from the linker 

6 - interrupt frequency need to be high and this is Ok when wifi is not doing anything, but as soon as there is TCP/IP exchange that is significative like a 100 chars - the ESP32 crash.
If no isr or they are 1 / sec (which is useless for stepper and for temperature control) the wifi is fine.
For what I can see pure Wifi is on Core 0, when Marlin is on Core1, I have moved the web server to core 0 creating a task and  this seems not improving the isr crash when TCP/IP and isr are working together
I have tested  using a sync web server, an async web server and pure wifi server none of them is working

I did also a test script with only one isr, which do nothing but enabling disabling the flag `in_temp_isr` and displaying big page or uploading file give same results :    Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
So issue is not in only Marlin but in usage of isr itself

I did not found how to assign isr to only one core and I do not know if it is the way to go 
but if Wifi cannot be used in same time as printer FW it is a dead end ...
I will try to have a chat to esp32 developers to get some insight of what could be a solution if any

simon-jouet, 2018-03-06:
Hi Luc,

Thanks very much for looking into this. You're right about getting some more insight into this if the WiFi isn't working and due to the limited GPIO it's a bit of a dead end.

My longer term idea was to replace the stepper ISR by the RMT module and therefore free up the CPU a lot and use a standard task instead of an ISR for the temperature. The problem is that both these changes are probably quite a significant amount of work and will diverge quite a bit from the standard marlin than having this merged in will be unlikely.

I'm curious if you hear anything from the ESP32 devs about this :)

Cheers

luc-github, 2018-03-06:
Hi Simon,
I have created a test sketch using isr and webserver example : https://gist.github.com/luc-github/c80746245c1f600ec5810f0785470263 To show the issue out of Marlin FW.

 I have shared it to arduino-esp32 chat (https://gitter.im/espressif/arduino-esp32?at=5a9d28a153c1dbb743a653c1) and got some preliminar private feedback from @me-no-dev the esp32 guru but no solution yet

luc-github, 2018-03-20:
I got suggestion from Chat : https://esp-idf.readthedocs.io/en/latest/api-reference/system/esp_timer.html
I have used it for another project (https://github.com/luc-github/ESP32SSDP/blob/master/ESP32SSDP.cpp#L416-L428), and it looks promising but I did not tested yet on Marlin sorry

luc-github, 2018-07-06:
close as isr code changed

closed 2018-07-06 by luc-github