simon-jouet/Marlin #10: Error compiling

iesfausti, 2018-02-28:
### Compile Error

Hi
I had to add/hack/fake the code below  to compile, none of those were found.
What am I missing?
(I just download your code and use arduino 1.8.3 plus 
git clone https://github.com/espressif/arduino-esp32.git esp32 )

#define MYSERIAL0 Serial
typedef uint32_t hal_timer_t;
#define NUM_SERIAL 1
#define STEPPER_ISR_ENABLED() 1//TEST(TIMSK1, OCIE1A)
FORCE_INLINE static hal_timer_t HAL_timer_get_compare(const uint8_t timer_num) {

}
#define PULSE_TIMER_NUM STEP_TIMER_NUM
#define PULSE_TIMER_PRESCALE STEPPER_TIMER_PRESCALE
FORCE_INLINE static void HAL_timer_set_compare(const uint8_t timer_num, const hal_timer_t compare) { /* ???? */
}

Thanks in advance

simon-jouet, 2018-03-06:
Hi @iesfausti this might be related to the merge @thinkyhead did against the branch, I haven't had time to have a look at this since then.

For the time being if you want to see that status of this you can just pull the version before the merge which should compile fine. I will get back into this once my new frame is ready to go

simon-jouet, 2018-06-03:
@iesfausti I have spent some time in the last week to work on the HAL, the latest code should compile fine using platformio or esp-idf as a component. (I haven't tried from the arduino IDE)

closed 2018-06-03 by simon-jouet