simon-jouet/Marlin #11: Test ESP32

3d-gussner, 2018-06-07:
### Description

Test ESP32 Marlin code

### Steps to Reproduce

Every print i try my https://wiki.wemos.cc/products:lolin32:lolin32 reboots

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Here the OctoPrint terminal results 

```
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
[...]
Changing monitoring state from "Connecting" to "Operational"
[...]
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:EEPROM:0
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:0
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: ok
Send: M21
Recv: echo:Unknown command: "M21"
Recv: ok
Send: M155 S2
Recv: ok
[...]
Changing monitoring state from "Operational" to "Printing"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M107*36
Recv: ok
Send: N2 M115 U3.1.0*99
Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:EEPROM:0
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:0
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: ok
Send: N3 M155 S2*96
Recv: ok
Send: N4 M83*28
Recv: ok
Send: N5 M104 S215*102
Recv: ok
Send: N6 M140 S55*83
Recv: ok
Send: N7 M190 S55*95
Recv: ok
Send: N8 M109 S215*102
Recv: ok
Send: N9 G28 W*109
[...]
Recv: echo:busy: processing
Printer seems to support the busy protocol, adjusting timeouts and setting busy interval accordingly
[...]
Recv: echo:busy: processing
Recv: X:-20.00 Y:0.00 Z:0.00 E:0.00 Count X:-2000 Y:0 Z:0
Recv: ok
Send: N10 M113 S2*80
Recv: ok
Send: N11 G80*33
Recv: echo:Unknown command: "G80"
Recv: ok
Send: N12 G1 Y-3.0 F1000.0*27
Recv: ok
Send: N13 G1 X60.0 E9.0  F1000.0*97
Recv: ok
Send: N14 G1 X100.0 E12.5  F1000.0*110
Recv: ok
Send: N15 G92 E0.0*109
Recv: X:100.00 Y:0.00 Z:0.00 E:0.00 Count X:-2000 Y:0 Z:0
Recv: ok
Send: N16 M900 K30*117
Recv: echo:Unknown command: "M900 K30"
Recv: ok
Send: N17 G21*44
Recv: echo:Unknown command: "G21"
Recv: ok
Send: N18 G90*41
Recv: ok
Send: N19 M83*32
Recv: ok
Send: N20 G1 E-0.80000 F2100.00000*15
Recv: ok
Send: N21 G1 Z0.600 F7200.000*52
Recv: ok
Send: N22 G1 X78.634 Y82.751 F7200.000*99
Recv: ok
Send: N23 G1 Z0.200 F7200.000*50
Recv: ok
Send: N24 G1 E0.80000 F2100.00000*38
Recv: ok
Send: N25 M204 S1000*96
Recv: ok
Send: N26 G1 F1800*115
Recv: ok
Send: N27 G1 X79.985 Y82.623 E0.04257*84
Recv: ok
Send: N28 G1 X170.004 Y82.623 E2.82246*111
Recv: ok
Send: N29 G1 X171.356 Y82.749 E0.04259*100
Recv: ok
Send: N30 G1 X172.666 Y83.121 E0.04268*98
Recv: ok
Send: N31 G1 X173.884 Y83.728 E0.04268*111
Recv: ok
Send: N32 G1 X174.979 Y84.556 E0.04304*111
Recv: ok
Send: N33 G1 X175.890 Y85.558 E0.04245*98
Recv: ok
Send: N34 G1 X176.609 Y86.723 E0.04292*111
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: ok
Send: N35 G1 X177.099 Y87.993 E0.04268*97
[...]
Recv: echo:busy: processing
Recv: ok
Send: N36 G1 X177.351 Y89.392 E0.04458*101
[...]
Recv: ok
Send: N37 G1 X177.338 Y120.753 E0.98330*92
Recv: ok
Send: N38 G1 X177.120 Y121.924 E0.03733*82
Recv: ok
Send: N39 G1 X176.875 Y122.676 E0.02482*88
Recv: ok
Send: N40 G1 X176.326 Y123.792 E0.03899*86
Recv: ok
Send: N41 G1 X175.884 Y124.449 E0.02482*82
Recv: ok
Send: N42 G1 X175.376 Y125.049 E0.02466*88
Recv: ok
Send: N43 G1 X174.451 Y125.883 E0.03904*92
[...]
Recv: echo:busy: processing
[...]
Recv: ok
Send: N44 G1 X173.798 Y126.323 E0.02470*87
Recv: ok
Send: N45 G1 X172.714 Y126.859 E0.03792*91
Recv: ok
Send: N46 G1 X171.366 Y127.249 E0.04399*88
Recv: ok
Send: N47 G1 X170.015 Y127.377 E0.04256*81
Recv: ok
Send: N48 G1 X79.901 Y127.375 E2.82545*97
Recv: ok
Send: N49 G1 X78.719 Y127.264 E0.03723*110
Recv: ok
Send: N50 G1 X77.876 Y127.063 E0.02716*109
Recv: ok
Send: N51 G1 X76.711 Y126.603 E0.03927*111
Recv: ok
Send: N52 G1 X75.637 Y125.947 E0.03947*96
Recv: ok
Send: N53 G1 X74.966 Y125.391 E0.02732*103
Recv: ok
Send: N54 G1 X74.129 Y124.466 E0.03911*99
Recv: ok
Send: N55 G1 X73.679 Y123.801 E0.02518*105
Recv: ok
Send: N56 G1 X73.141 Y122.714 E0.03804*109
Recv: ok
Send: N57 G1 X72.751 Y121.370 E0.04387*104
Recv: ok
Send: N58 G1 X72.623 Y120.034 E0.04209*102
Recv: ok
Send: N59 G1 X72.657 Y89.292 E0.96388*91
Recv: ok
Send: N60 G1 X72.878 Y88.084 E0.03852*86
Recv: ok
Send: N61 G1 X73.123 Y87.331 E0.02482*83
Recv: ok
Send: N62 G1 X73.448 Y86.615 E0.02466*80
Recv: ok
Send: N63 G1 X74.109 Y85.560 E0.03903*91
[...]
Recv: echo:busy: processing
[...]
Recv: Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Recv: Core 1 register dump:
Recv: PC      : 0x40085513  PS      : 0x00060e34  A0      : 0x800de7fd  A1      : 0x3ffc0b20
Recv: A2      : 0x3ffc10d8  A3      : 0x3ffc2c75  A4      : 0x3ffc0b48  A5      : 0x3ffc10d8
Recv: A6      : 0x3ffc2e8c  A7      : 0x00000001  A8      : 0xb33fffff  A9      : 0x0000abab
Recv: A10     : 0xb33fffff  A11     : 0x0000cdcd  A12     : 0x800d70f5  A13     : 0x00000001
Recv: A14     : 0x00060e23  A15     : 0x80000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000006
Recv: EXCVADDR: 0x00000000  LBEG    : 0x400e5802  LEND    : 0x400e5839  LCOUNT  : 0x00000000
Recv: Core 1 was running in ISR context:
Recv: EPC1    : 0x40080e9c  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40085513
Recv: 
Recv: Backtrace: 0x40085513:0x3ffc0b20 0x400de7fa:0x3ffc0b40 0x400d25c3:0x3ffc0b70 0x400d8d52:0x3ffc0ba0 0x400d8da7:0x3ffc0bd0 0x40080e89:0x3ffc0bf0 0x40081695:0x3ffc0c10 0x40080e99:0x00000000
Recv: 
Recv: Core 0 register dump:
Recv: PC      : 0x400e235a  PS      : 0x00060734  A0      : 0x80085aa8  A1      : 0x3ffc8fe0
Recv: A2      : 0x00000008  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffc9624
Recv: A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc3be4  A9      : 0x3ffc3bc8
Recv: A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000001
Recv: A14     : 0x00060021  A15     : 0x00060e23  SAR     : 0x00000000  EXCCAUSE: 0x00000006
Recv: EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Recv: 
Recv: Backtrace: 0x400e235a:0x3ffc8fe0 0x40085aa5:0x3ffc9000
Recv: 
Recv: Rebooting...
Recv: ets Jun  8 2016 00:22:57
Recv: 
Recv: rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
Printer requested line 0 but no sufficient history is available, can't resend
Changing monitoring state from "Printing" to "Error: Printer requested line 0 but no sufficient history is available, can't resend"
Changing monitoring state from "Error: Printer requested line 0 but no sufficient history is available, can't resend" to "Offline (Error: Printer requested line 0 but no sufficient history is available, can't resend)"
Connection closed, closing down monitor
Closing down send loop
```

simon-jouet, 2018-06-07:
Hey @3d-gussner thanks for having a look at this, good to see someone else trying it out :)

The issue you're having is the same one I'm a bit stuck on that i posted last night https://github.com/MarlinFirmware/Marlin/issues/6524#issuecomment-395143930.

I spent some time on this trying to figure it out in the last few evenings and it seems to be when acquiring the locks for the timer but I'm not too sure. I might try to plug-in my ft2232h and use openocd to try and debug that...

If by any chance you figure it out that would be great too :D

3d-gussner, 2018-06-07:
I'm not that good in debugging but found something that may help if I am not wrong https://github.com/espressif/arduino-esp32/issues/855
Hope that helps

simon-jouet, 2018-06-09:
@3d-gussner just pushed some code which "solved" the WDT timeout issue -- you can try that out if you want. I managed to print 3/4 of a calibration cube :) not too sure why the end failed just yet but at least it's progress!

3d-gussner, 2018-06-11:
@simon-jouet Thanks, wait for the serial fix @thinkyhead https://github.com/MarlinFirmware/Marlin/issues/6524#issuecomment-396102710 mentioned is included.
BTW: which board do you use to test print? Your https://github.com/simon-jouet/ESPRamps ?

simon-jouet, 2018-06-11:
Hey @3d-gussner, I've rebased yesterday evening but few more things have been merged since; I will do another rebase in the next few hours :)

For the board I'm not using the espramps one as it had some issues with the pinout. For the time being I just soldered everything on a proto board so I can try things out. Once I have a more definite pinout I will make a all in one pcb instead of using a ramps board

simon-jouet, 2018-06-11:
Okay it's rebased to the latest bugfix-2.0.x now

closed 2021-12-24 by 3d-gussner