simon-jouet/Marlin #12: Start from beginning

faronov, 2019-04-02:
I have some question regarding this cool project
1. this branch still actual? or everything that needed for running esp32 as controller already exists in official marlin branch?
2. I try out official Marlin branch, it compiles without a problem and I also able flash this binary to dev board but big question if  I upload code to some esp32 dev board like "DOIT Esp32 DevKit v1" project will run without a connection to additional hardware?
for my test, it's just crashed and I try to understand this because some hardware must be connected and if it's missing I must expect that it crashes?
I like the idea to use esp32 as the controller board

if it's expected that devices won't' boot without additional hardware what is the minimum required hardware?
I plan to create some minimal with SPI drivers 
And want start from some minimal hardware connected to this board

faronov, 2019-04-02:
I only see

faronov, 2019-04-03:
Looks like I have low power for running esp + wifi enabled from USB
removing USB support helped to run Marlin official build successfully
this branch has an error in building step
Marlin\src\module/stepper.h:150:102: error: no matching function for call to 'max(long int, long unsigned int)'

luc-github, 2019-04-03:
@faronov Simon code is already merged to Marlin 2.x - and Marlin is more up to date
Also be sure you use latest ESP32 core - I mean latest GIT not the released one
Also even there is a basic WebUI in marlin 2.x you may want to test the port of ESP3D for Marlin : https://github.com/luc-github/Marlin/tree/bugfix-2.0.x

faronov, 2019-04-08:
How I can be sure that I use the latest esp32 core in Platformio?

luc-github, 2019-04-08:
I do not use PlatformIO so I cannot be sure but reading http://docs.platformio.org/en/latest/platforms/espressif32.html?utm_source=github&utm_medium=arduino-esp32#using-arduino-framework-with-staging-version

Stage version should be uptodate as GIT was updated one month ago    
If it can help for checking : on latest GIT `ESP.getSdkVersion()` give me __v3.3-beta1-506-gebdcbe8c6__