simon-jouet/Marlin #13: TMC2209

simon-jouet, 2020-03-28:
@felixstorm @vivian-ng

(Decided to post here instead of the main repo because I assume that's an issue only for ESP32 users.)

Hey guys, hopefully you get a notification from those mentions :), with what has been happening I finally took some time this weekend top bump my Marlin version and trying to do some minor updates to my printer.

I got TMC2209 a while back (fystec v2.1) and i'm running into some issues with the configuration, so I was hoping one of you had a try. Setting up a single TMC2209 works perfectly fine, the stepping works okay and doing a M122 returns the status of the driver as expected.

The problem is when I start enabling more that one TMC2209 for the same UART, if I enable both X and Y, give X the slave address 0 and y the slave address 1 then the stepping stops working. The driver is enabled (the stepper is on hold), M122 returns the status of all the drivers but stepping just doesn't do anything.

If I comment out the driver type from tmc2209 for either X or Y (so only a driver is tmc2209) then the one configured for tmc2209 is working perfectly, just can't have more than one TMC2209 at a time.

Do you have any clue what could be going on before I start digging too deep into this?

Here is the diff of my branch if that helps
[esp32.txt](https://github.com/simon-jouet/Marlin/files/4397689/esp32.txt)

vivian-ng, 2020-03-29:
@simon-jouet I did my initial test on TMC2209 using X and Y too with no issues... anyway, right now, I am using TMC2209 on X, Y, Z, and E0.

`Configuration.h` set as:
```
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209
```

`Configuration_adv.h` set as:
```
  #define  X_SLAVE_ADDRESS 0
  #define  Y_SLAVE_ADDRESS 1
  #define  Z_SLAVE_ADDRESS 2
  #define E0_SLAVE_ADDRESS 3

  #define X_HARDWARE_SERIAL Serial1
  #define Y_HARDWARE_SERIAL Serial1
  #define Z_HARDWARE_SERIAL Serial1
  #define E0_HARDWARE_SERIAL Serial1
  #define HARDWARE_SERIAL1_RX 21
  #define HARDWARE_SERIAL1_TX 22
  //#define MONITOR_DRIVER_STATUS
  #define TMC_DEBUG
```

Can I check which version of Marlin are you using? I think pins 25 and 26 are being used for I2S on your original `pins_esp32.h` so I don't know if your configuration to use these pins for TX and RX could be an issue.

My last compile was on March 24, using bugfix-2.0.x pulled on March 23. No issues with moving X, Y, and Z axes, will test out a print later if necessary.

simon-jouet, 2020-03-29:
Thanks @vivian-ng, good to know that it's working well. I will try it again today now that i've got time to think it over and know that it should work :)

I'm using latest bugfix 2.0.x branch from yesterday (commit 2f7e5cf7ddd04ae0474f104dd1b712b480fe692c)

Regarding the pins you're right but the I2S version of my controller uses different pins, just that the config on marlin was never updated for the i2s version of my board, considering stepping and status work when only one driver is selected I don't think it's the issue.

felixstorm, 2020-03-30:
@simon-jouet Yes, your mention did reach me and nice to hear that you are still using the ESP32 with Marlin. My board has now been in use for almost a year without ever wanting to go back 😉  
Unfortunately I only use TMC220**8** drivers in standalone mode (not even connected via SPI), so there's probably not much I can help you with 😞

simon-jouet, 2020-03-30:
@felixstorm Yeah still using my ESP32 board, I just had (and have) little time to play with my printer in the last 6 months or so. It was quite a step up when I tried to update my Marlin version this weekend :). Using Luc's webui is a significant improvement (I might play with the javascript/html at some point).

Thanks for the details regarding your setup, I've been tempted to just put my 2209 in standalone mode, but I'm in no rush to get that plugged in (still using drv8825 on the printer at the moment). And considering it's working for @vivian-ng it's probably because i'm doing something wrong, although with even more debugging yesterday I didn't find what the issue is...

Thanks both for the reply, I hope everyone is keeping well with what has been happening

vivian-ng, 2020-03-30:
@simon-jouet Try taking a look at this [blog post](https://teck78.blogspot.com/2020/01/setting-up-tmc2209-in-uart-mode-on-mrr.html), it talks about the wiring for using TMC2209s on a single UART through slave addresses. The part about the hardware serial `#define`s are outdated, though.

simon-jouet, 2020-03-30:
Thanks @vivian-ng I will look through this in more details maybe tonight, but from the looks of it just now the wiring looks to be okay (I used different tmc2209 drivers) but nothing is obviously wrong... :(

simon-jouet, 2020-03-30:
@vivian-ng still no success, what's in the blog is pretty much exactly what I do. I use fysetc 2.1 drivers that have the UART config a bit different but the UART bit looks to be working so I don't think that's the issue.

I've basically got the following and it doesn't work, if I comment out X_DRIVER_TYPE or Y_DRIVER_TYPE (to only have a single TMC2209) then it works fine.

```
#define X_DRIVER_TYPE TMC2209
#define Y_DRIVER_TYPE TMC2209

#define  X_SLAVE_ADDRESS 0
#define  Y_SLAVE_ADDRESS 1

#define MONITOR_DRIVER_STATUS
#define TMC_DEBUG
```

I just had a look with the logic analyzer and I can see that when only one TMC2209 is present then the steps are sent properly, however with 2 TMC2209 the steps are not sent. Now I have to figure out why they are not sent... need to figure out first if it's a hardware or software issue...

vivian-ng, 2020-03-31:
@simon-jouet Try printing with other drivers, if they work, then I think we can assume the I2S stepper stream is working fine, and it is more likely something else (firmware or hardware) that is causing the problem. If you have changed the pins since the last time, it may be a good idea to double check that all the pins are correctly assigned in the pins file with respect to the actual hardware. I made that mistake before when I redid my board but forgot to update the pins file...

simon-jouet, 2020-03-31:
@vivian-ng I started working on this last night, it works fine with other drivers and it works fine with TMC2209 too (but just when one is used). If i use more than one tmc2209 (regardless of the axis) it stops working, if I try every axis independently it works...

I've started debugging a bit, my guess is that something is wrong initializing the stepper stream for some reason. I haven't had much time last night but I will try to look at that this week

simon-jouet, 2020-03-31:
Okay I've figured it out, it's a bug in the code, not exactly sure what's the exact cause of the issue but I've moved `i2s_init()` from HAL_init() to the end of `HAL_init_board()` and it's working.

My guess the cause is that Serial2 is somehow initialised with multiple drivers therefore the GPIO17 is redefined from being a I2S pin to be a UART pin and the stepping fails.

Looking at the code I can see that the BOARD_INIT() is done after the TMC Serial bit, so I would guess that's why. Going to look a bit more, to see if I can open a proper MR on Marlin instead of just moving code around

simon-jouet, 2020-03-31:
Thanks guys :), I've opened a MR to "fix" that

https://github.com/MarlinFirmware/Marlin/pull/17351

closed 2020-03-31 by simon-jouet